Seeed-Studio/Mutichannel_Gas_Sensor #1: Work with Raspberry pi 2

rbarretto18, 2015-10-08:
Sir,

I have connected the Grove multichannel sensor to the I2C port on my Grove pi + board which is attached to my Raspberry pi 2. I have saved the latest library in sketchbook folder. From my Grove Raspbian wheezy OS when I open Arduino 1.0.1 IDE and open the file from File --> Sketchbook --> Grove Multi channel sensor and verify it, I get a compilation error.
 Is it because I'm using a Raspberry pi?
 Is there another way to work it with Raspberry Pi. If I develop a script in Python would it work or would I be wasting my time as the device is incompatible. Thanks in advance.

Richard

awong1900, 2015-10-09:
Hi @rbarretto18 
Can you post the error here?

rbarretto18, 2015-10-09:
Sir,

The error is 

ReadSensorValue_Grove.cpp:16:34: fatal error:
MultichannelGasSensor.h: No such file or directory
compilation terminated.

In my Sketchbook folder i have the following folders and files
Slave(folder)
Example(folder)
MultichannelGasSensor.cpp
License.txt
README.md
MultichannelGasSensor.h

KillingJacky, 2015-10-09:
You should place the library under Arduino IDE's libraries directory rather than the sketchbook directory, and open the example project from file->example

rbarretto18, 2015-10-09:
Sir,

Thank you..I did it and I was able to compile and verify successfully. But now when I chose Upload it gives me an error

Problem uploading to board
avrdude: stk500_recv(): programmer is not responding

Under TOOLS the BOARD selected is Arduino Uno although my Raspberry Pi is connected to a Grove+ board
SERIAL PORT is /dev/ttyS0
PROGRAMMER is not selected yet.
The options for PROGRAMMER are
AVR ISP
AVRISP mkll
USBtinyISP
USBasp
Parallel Programmer
Arduino as ISP

Im not sure which programmer will work since Im using a raspberry pi 2 instead of arduino or should I change the board options

KillingJacky, 2015-12-07:
Have you tried every type of the programmer? Have you get over this problem?

rbarretto18, 2016-02-24:
Yes I tried every type of programmer but doesnt work. I have a Raspberry Pi
2

closed 2024-10-15 by baorepo