Seeed-Studio/Mutichannel_Gas_Sensor #2: Calibration

Roysten, 2015-12-06:
Should this sensor be calibrated when it is fully heated? I have calibration gas (150ppm NO) but I am not able to get the reported values even close to 150ppm.

KillingJacky, 2015-12-07:
Hi Roysten,

The sensor should be calibrated at least once when moved to new environment. Then the calibration state can be store in the module's flash storage.

Roysten, 2015-12-15:
Thank you for your reply. Should the sensor be calibrated when the sensor is fully heated or when it is still cold? How long does it take the sensor to reach the optimal temperature? I have let it warm up for several hours but the measurements where still ten times too small (after calibration of course).

KillingJacky, 2016-02-24:
Hi Roysten,
Sorry for late reply, cauz your question is pretty tough. I'm not sure whether the sensor is so accurate to meet the industrial measurement's requirement. We have forward your question to the original of this sensor, but no good returned. What result did you investigate in the following efforts?

Roysten, 2016-02-24:
I'm fairly certain the sensor is not accurate enough to measure a concentration of 100 ppm. It does work though, I've tried measuring the concentration of CO using the exhaust of two different cars:
old BMW:
[image]
Volkswagen Lupo:
[image]
And the last measurement I have done was running the BMW in a closed garage:
[image]

So it does register CO, but definitely not in the range of 0 ~ 100 ppm. Since I do not have access to other calibration gases I unfortunately cannot calibrate the sensor.

KillingJacky, 2016-02-24:
Thanks for your investigation! I'd be sorry to say that this sensor disappoints us when it works under low concentration environment. Its price is pretty high but sadly it's still not competent for accurate measuring.

aldoagosta, 2016-03-03:
Hi Roysten,

here you will find some info on how to calibrate the sensor that is supposed to be after warming phase.
https://www.cdiweb.com/datasheets/e2v/AN172_Useandperform.pdf

Which sketch did you use for make it working on Arduino? Mine works for NO2 only if applying the package provided by SeedStudio.

Thank you

Aldo

Roysten, 2016-03-03:
@aldoagosta I used the code that is available in this repository. Indeed it does not seem to measure concentrations in the specified ranges for atleast CO, so I guess that your concentrations are too high/low. Also keep in mind that the heating can take quite a while.

aldoagosta, 2016-03-04:
Hi Roysten, thanks. 

I tried the sensor after weeks it was turned on then I do no think it can be a matter of heating.

The strange fact is that as you put ammonia beside the sensor the related value stands still while the NO2 decreases quite fast. For NO2 is very sensitive at least.

Moreover even being in 2 different places and countries the values mpierre and I read in the serial monitor are identical for NH3 (0.99) and CO (1.20) while NO2 varies.

Thank you 

Aldo

reisod, 2016-03-31:
I'm in the same boat.  I bought three with the need for measuring NH3 and CO.  I can take a 50 ppm NH3 source, and still not get any deviation from the 0.99 NH3 reading.  I'm also flat lining at 1.20 CO.  It's not you - it's the $50 sensor board that does not do what it claims to do.

aldoagosta, 2016-04-01:
Hy Roy,

we are in three experiencing the same issue and getting identical values:
0.99 NH3, 1.20 CO etc.

Please, place a claim here:

http://www.seeedstudio.com/depot/index.php?main_page=support

They are even kind but I am still waiting for a solution since last
February.

My guess is there's something wrong in the electronics around the sensor
(it's not the sensor itself).

Have a nice day

Aldo

alejoduque, 2017-05-09:
hello all, i have one of this sensors also but haven't been able to address it to the correct I2C grove connector on a Seeeduino LoRa board that has 4 inputs. I will like to try if I get the same readings and support your search and will ping back once I get it to send something over the serial port.. so far nothing appears using the default I2C address of the slave it 0x04... there's no documentation about it.

awong1900, 2017-05-09:
@alejoduque http://wiki.seeed.cc/Grove-Multichannel_Gas_Sensor/  Hope help you.

alejoduque, 2017-05-09:
Thanks Ten but the question is how to address it through the Seeeduino LoRaWan / GPS board?
I have tried: 
    gas.begin(0x04);//the default I2C address of the slave is 0x04

But it is not found there on my board and can't find detailed documentation on other board with multiple grove inputs.

Please advice

awong1900, 2017-05-09:
Can you open a new issue about this?

alejoduque, 2017-05-09:
here you go: https://github.com/Seeed-Studio/Mutichannel_Gas_Sensor/issues/7

AshuJoshi, 2017-05-24:
I am getting very high readings for CH4 / methane - very, very high - with Arduino Uno Rev 2 and the Grove Shield. All others seem to be ok. I tried running calibration but it never ended if keeping it running overnight.

aldoagosta, 2017-12-27:
I tried to calibrate the sensor as well with new firmware and libraries. Calibration never ends even after 6 hours. The serial monitor report the restistance values (I guess) only but the process never ends.

closed 2024-10-15 by baorepo