Seeed-Studio/Mutichannel_Gas_Sensor #3: xadow with pic

vish287, 2016-01-21:
I am using the xadow multichannel gas sensor with pic microcontroller using FPC DIP addapter using 0.5mm FPC. I have converted this ardiuno code.

I communicate with it using software I2C and using address 0x04 for device address.  I have certain queries . please guide me.

1) when I use device ID 0x04, I use 0x08 for write and 0x09 for read
 I get LED D1 on when I send command 0x21 for power on as
i2c_xadow_start();
    i2c_xadow_sendbyte(0x08); // device address
    i2c_xadow_sendbyte(0x21); // to start the sensor heater
    i2c_xadow_stop();
    delay_ms(10);

2) now when I change device address using command 0x23

so Device Id 0x04 to 0x10
Write command 0x20
and read 0x21

power on is working
I again change the Device ID to 0x04
so communication to device is working...
3) now as provided by your ardiuno code, when I write code to simple read the 0x11 resister for res0[0],  I should get buffer[0] to buffer[3] values. I have taken buffer[4] as unsigned char but I am getting 0xFF (255)

my code for read data is as under

long readData(unsigned char addr)  // int
{
    //int timeout = 0;
     unsigned char buffer[4];
    long checksum = 0;// int? // unsigned char?
   long rtnData = 0;  //int?

    i2c_xadow_start();
    i2c_xadow_sendbyte(0x08); // default device address
    i2c_xadow_sendbyte(addr);  // give the address which is to be read
    delay_ms(50); // change delay if needed
    i2c_xadow_start();
    i2c_xadow_sendbyte(0x09); // read the device
delay_ms(50);
    buffer[0] = i2c_xadow_readbyte(0); // read the four bytes
    buffer[1] = i2c_xadow_readbyte(0);
    delay_ms(50);
   buffer[2] = i2c_xadow_readbyte(0);
    buffer[3] = i2c_xadow_readbyte(1);
    i2c_xadow_stop();
delay_ms(50);
    checksum = (long)(buffer[0]+buffer[1]+buffer[2]);
//    if(checksum != buffer[3])
//    {
//        return -4;  //
//
//
//    }
//    else
//    {
    rtnData = ((buffer[1] << 8) + buffer[2]);
   // }
  return (rtnData);//successful
}

please guide me why I am not getting values for read operation.

KillingJacky, 2016-01-22:
Hi, 
I think you should check your implementation of i2c_xadow_readbyte function, or even all the soft i2c related functions. I noticed that you might mix up the i2c device address and the register address inside a i2c device. Additionally, the default i2c address of this sensor 0x04 is a 7 bits i2c address. 7 bits i2c address has no sense of the increase between read and write.

vish287, 2016-01-23:
yes sir. I have checked this functions with other sensor BMP180 and it is giving me all the parameters with this soft I2C. now as it is 7 bit address as 0x04 so using write it will be 0x08 and for read 0x09. as I2C protocol, I am sending device address, then register address and device address with read as 0x09.

Here for trial base i have take delays of 50msec. in your ardiuno code you have just taken delay_(2).
Now "7 bits i2c address has no sense of the increase between read and write." how it will work here?   here with I am attaching my .c file please guide me either it is data type issue or any other.
thanking you.
[code.docx](https://github.com/Seeed-Studio/Mutichannel_Gas_Sensor/files/101857/code.docx)

KillingJacky, 2016-01-25:
Hi vish,

Looking into the arduino code
```c
int16_t MutichannelGasSensor::readData(uint8_t cmd)
{
    uint16_t timeout = 0;
    uint8_t buffer[4];
    uint8_t checksum = 0;
    int16_t rtnData = 0;

    //send command
    sendI2C(cmd);
    //wait for a while
    delay(2);
    //get response
    Wire.requestFrom(i2cAddress, (uint8_t)4);    // request 4 bytes from slave device
    while(Wire.available() == 0)
    ......

void MutichannelGasSensor::sendI2C(unsigned char dta)
{
    Wire.beginTransmission(i2cAddress);               // transmit to device #4
    Wire.write(dta);                                    // sends one byte
    Wire.endTransmission();                             // stop transmitting
}
```
There's a STOP  between address writing transmission and the following reading transmission. So I guess you should issue a STOP here:

```c
![image](https://cloud.githubusercontent.com/assets/5130185/12541736/aa42a0c2-c355-11e5-8fc6-c29a27f00a07.png)
```

The simulated I2C slave might not support repeated START operation. Please give a try with the above workout and feedback with any result. 

Thank you.

vish287, 2016-02-04:
Hi,
I want to know in the code how the ratio range is decided? as from data sheet its difficult to get the exact ration range of RS/R0 for calculations. Thanks

KillingJacky, 2016-02-14:
Hi,
The ratio is decided from the curve in the datasheet, there's no exact formula but we simulated one with sample points extracted from the curve.

closed 2024-10-15 by baorepo