Seeed-Studio/Mutichannel_Gas_Sensor #4: Sensor seems abel to read NO2 value only

aldoagosta, 2016-02-27:
Hi,

it seems that sensor (Grove version of MiCS-6814) does not read concentrations other than NO2.
NO2 is the only value changing over time while NH3 stands still even with a bottle of Ammonia placed below. By the way putting the ammonia bottle beside the sensor NO2 level decrease while NH3 stands still as said above.

The sensor is connected to a I2C port on arduino one board thru Base Shield 2.

Thank you for support.
Regards

Aldo

empierre, 2017-09-18:
I am waiting too for an answer or is this sensor crap ?

loovee, 2017-09-18:
There're 3 independent channel for the sensor MiCS-6814, RED sensor, OX sensor as well as NH3 sensor. 
Please refer to this [datasheet](https://raw.githubusercontent.com/SeeedDocument/Grove-Multichannel_Gas_Sensor/master/res/MiCS-6814_Datasheet.pdf). 
Thanks.

closed 2024-10-15 by baorepo