Seeed-Studio/Mutichannel_Gas_Sensor #5: No2 Reading and Drift

darkstarlabs, 2016-03-17:
I have two of these modules on the same bench setup on two different 2560's collecting data for No2 every 5 minutes using mutichannelGasSensor.measure_NO2();

Currently one is reading 2.58 and the other 3.43

When first powered up, the reading after 5 minutes is 0.15, 0.59 after an hour, 1.78 after a day, and now at 2.58. They both seem to follow the same drift and its unlikely that the No2 concentration in the lab is anywhere near 3.43.

Should I be powering these down regularly or leaving them powered?

You also have a doCalibrate(void) but Its not clear if this should be used, and what effect it may have? Do you have the source for the onboard MP? 

Any help would be greatly appreciated

KillingJacky, 2016-03-21:
Hi,
It seems that the sensor faces common drifting and bad accuracy issues. We're talking with the original and driving them to investigate on this.
And your question, " Should I be powering these down regularly or leaving them powered? "
I should say that powering them up constantly will be better, because this sensor should be working under a warmed up env. The more stable the env temperature be the more accurate result it gets.

It has no need that doCalibrate be called regularly, but the advice for the serious calibration is, warm the sensor up for a longer time (e.g. one minute or more) and then call the doCalibrate. After doCalibrate is done, the calibrated parameters (reference values for clear fresh air) will be stored in the flash.

https://github.com/Seeed-Studio/Mutichannel_Gas_Sensor/blob/master/slave/Slave_Grove.ino is the code running on the onboard MP.

darkstarlabs, 2016-03-22:
Thanks for the info. When you say reference to 'clean fresh air' in doCalibrate() what exactly is clean fresh air? We're in a city, the air is never clean or fresh. I suppose I could place the unit in a high vacuum and then fill with argon to ensure there is zero No2 but I'm not even sure if the sensor will survive absolute vacuum. I suppose I could place it in a stream of Argon under pressure and then calibrate it, but it seems a little excessive. The No2 response curve is a strait line, and there doesn't seem to be any temperature compensation required so I can't see why its so far out, unless maybe the sensors are damaged by the reflow process? I do believe CDA have a socket for this sensor (and the other sensors in the MiCS range, maybe I'll order some of those, and some fresh sensors and drop a socket on there if you've not already looked at the possibility of sensor damage from reflow?

KillingJacky, 2016-03-23:
I mean the fresh air is the air having no much target gas like No2. I think for high accurate measurement the calibration process will be tedious. So the reference for this sensor is just the same env as the normal air condition in wide open place. We're very sorry for the inconvenience and actually we haven't much stratagem to verify whether the sensor is damaged during reflow process because the target gas isn't handy to have. It seems we have to go to the original's lab to perform the verification and test.

darkstarlabs, 2016-03-26:
Ok, so I've ordered some MiCS-6814 sensors from CDX along with their prototype socket and I'll see if I can socket one just to see. I'm also going to try calibrating it in a zero no2 atmosphere and then read the resistances at that point (and also from a reference sensor that's socketed). Its the drift that concerns me so I'm also going to stick a PTFE screen over one to see if something is polluting the sensor or if ambient air currents are causing temperature variances in the heater - either worth a check. 
Speaking of sensors, where can I get more of the boards? Seems like grove have no stock anymore. I'm going to need another two or three to figure it out.

reisod, 2016-03-31:
Robotshop had them for a bit.  There's a few on Amazon.  I'm weary about the board though, I have three and none of them are actually giving usable readings.

darkstarlabs, 2016-04-03:
The sensor is good as the MiCS sensors are used everywhere. I've got quite a few on the dev boards and all work fine once calibrated which leaves us with a few options - heater, ambient pollution, reflow damage or something wrong with the circuit. I'll work to figure it out but if anyone else figures it out before me please post here :)

darkstarlabs, 2016-04-04:
Ok, so here it is. Looking only at the NOX sensor, according to the circuit, 3.3v is supplied to a 56k resistor, then through the sensor to ground. The MCU takes its reading from between the two resistances, which is fine, except.. the voltage over the sensor is 1.61611v and the supply is 3.2704v so in order to get 1.61611v the resistance of the sensor must be also around 56k which is way outside the spec sheet of 20k max. The heater btw is 1.5370v at 23mA which is low but within spec. 

Anyone got any ideas?

aldoagosta, 2016-05-16:
See the doc hereafter where the sensor manufacturer specifies how to calculate values from sensor (does not matter if the models listed are different): 
https://www.cdiweb.com/datasheets/e2v/Load%20resistance%20for%20MiCS%20sensors%20measurements.pdf

manufacturer instructions here:
https://www.cdiweb.com/datasheets/e2v/AN172_Useandperform.pdf

BTW SeedStudio sensor implementation seems able to read NO2 values only. It does not work at all for NH3 and CO.

Gook luck!

Aldo

closed 2024-10-15 by baorepo