Seeed-Studio/Mutichannel_Gas_Sensor #7: How to address it when using multiple grove sensor boards

alejoduque, 2017-05-09:
The basic tutorial says:
    gas.begin(0x04);//the default I2C address of the slave is 0x04

But what if you are using a Seeeduino LoRaWAN board that has 4 inputs? How can I tell what input # is the one located to the outside of the board and what # will have the innermost? There's no clear documentation about this and I will like to use also an OLED and Humidity sensors but have no clue as on how to address them.

Please help with more documentation.

Thanks in advance

awong1900, 2017-05-09:
Here is LoRa wiki http://wiki.seeed.cc/Seeeduino_LoRAWAN/ . Read this
[image]

I do not fully understand what you mean. But if you want use Grove connector with Gas_sensor.
1. use digitalWrite(38, HIGH) to open VCC, 
2. plug  Grove on I2C port.  
3. Write your code to read data through i2c.

alejoduque, 2017-05-09:
Thanks, will try it.

awong1900, 2017-05-17:
@alejoduque Do you have try it?