Seeed-Studio/Mutichannel_Gas_Sensor #8: Reporting Wrong Values - especially for CH4

AshuJoshi, 2017-05-25:
I have been using the sample code to read values from the sensor on Arduino Uno with a Grove Shield. While the code works - I am not sure if the sensor is working or not. I know it is NOT accurate BUT the values reported are clearly crazy. For example some of the CH4 readings I am getting at my desk where I am experimenting with the sensor are:

2364214.25
2456180.25

I tried calibrating the sensor BUT the sensor would NOT end the calibration process even when I kept the code running overnight.

LPFraile, 2017-06-06:
I have exactly the same problem. My readings were not correct and because i had the firmware V1 i upgrade the firm version on the Mutichannel gas of Grove, and now I achieve the exactly same issue than you. I tried to calibrate the sensor too and  never END the calibration. I modify on MutichannelGasSensor.cpp the method doCalibrate(void) to  go out of the loop  while(1) in which is stack on. I think that is a library mistake. If you over pass this then the calibration data are update. Later of this my multichannel gas senor give correct values. 
Will be nice if someone can confirm if this stack on the loop is due to a mistake of the library program or is something else.  
Why to check this on the calibration?
  if((a0 - get_addr_dta(CH_VALUE_NH3)) > 2 || (get_addr_dta(CH_VALUE_NH3) - a0) > 2)cnt++;
                if((a1 - get_addr_dta(CH_VALUE_CO)) > 2 || (get_addr_dta(CH_VALUE_CO) - a1) > 2)cnt++;
                if((a2 - get_addr_dta(CH_VALUE_NO2)) > 2 || (get_addr_dta(CH_VALUE_NO2) - a2) >2)cnt++;

aldoagosta, 2017-08-26:
same issues here. both cray values and never ending calibration. 

@LPFraile could you please post the changes you applaied in .cpp? Thnak you!

Aldo

empierre, 2017-09-18:
Closed or solved ?

loovee, 2017-09-18:
Sorry just made a mistake. 
The calibration will wait until the value is stable. It may need minutes to half an hour. 
If you got never ending, try below:

1. Put the sensor to fresh air, maybe outdoor is better.
2. Change some paras in the library, in line 456, 457, 458 of .cpp file, change all of the 2 to larger number, such as 4. 

Thanks.

paulvha, 2017-09-27:
The calibration works fine for me, as long as I let the sensors warm up for at least 60 minutes. 

What VCC voltage are you applying and is that stable? The ADC is using the VCC input to calculate the ADC reference steps/voltages. If that is not stable, your reading will not be stable.
Put a glass over the sensor (when in clean air). At least you minimize that  the changes in reading is not because of changes air condition.
Next to that you can print the values of the different sensor while in the loop. Maybe one sensor is causing the issue that is causing the error count.

jakesmunx, 2018-01-31:
So was re-calibration the answer? The modules are factory calibrated, how can the values be so far out?

AndriyHz, 2018-04-29:
Hello dear friend!
Is it the effective sensor for measure outdoor air quality? Especially have interest in NO2 and CO measuring in public project.
Thank you.

aldoagosta, 2018-04-29:
Hi, mine is working since January. It is placed outside and I had not
issues so far. Unfortunately I cannot ensure about the measurements because
I never had the chance to compare them against other sensors.

If you want you can refer to this page for getting an example of how it
works.

https://thingspeak.com/channels/390635

Regards

Aldo

AndriyHz, 2018-04-29:
Hi @aldoagosta. Thank you for quick responses. 
I will try to use this sensor in my project (https://thingspeak.com/channels/371428).
Thanks.

AndriyHz, 2018-05-22:
Without NH4OH (indoor air)
The concentration of NH3 is 0.42 ppm
The concentration of CO is 2.40 ppm
The concentration of NO2 is 0.35 ppm
The concentration of C3H8 is 274.31 ppm
The concentration of C4H10 is 213.89 ppm
The concentration of CH4 is 67.34 ppm
The concentration of H2 is 0.29 ppm
The concentration of C2H5OH is 0.73 ppm
----------------
With NH4OH
The concentration of NH3 is 2.21 ppm
The concentration of CO is 74.23 ppm
The concentration of NO2 is 0.16 ppm
The concentration of C3H8 is 3380.55 ppm
The concentration of C4H10 is 1804.40 ppm
The concentration of CH4 is 22208746.00 ppm
The concentration of H2 is 54.83 ppm
The concentration of C2H5OH is 67.19 ppm

Why increase concentration CO, C3H8, C4H10, H2, C2H5OH but not only CH4, NH3?

sairamanujan, 2018-10-09:
Hello, 

Even I am not getting accurate values with grove sensor, I have performed 60 min calibration code outdoor. I am powering via USB supply.

Also I am not able to understand how do I confirm that these are accurate values or not. When I check with local air quality gas parameters, (Ref: http://airpollutionapi.com/aqi/india/telangana/kondapur)

Below are the values I am getting, 
The concentration of NH3 is 0.52 ppm
The concentration of CO is 4.23 ppm
The concentration of NO2 is 0.31 ppm
The concentration of C3H8 is 381.29 ppm
The concentration of C4H10 is 282.90 ppm
The concentration of CH4 is 554.21 ppm
The concentration of H2 is 0.69 ppm
The concentration of C2H5OH is 1.55 ppm

The above values obtained from sensor are around 10 times the actual values.

I am not able to find out the problem, @paulvha please help me out in finding out the issue

Thanks in advance

paulvha, 2018-10-09:
It has been some time since I work on this sensor:
couple questions : 
what platform (arduino/ESP8266/ Raspberry / ? ) do you use ?
which software ?
what is the VCC (5V or 3.3V)

sairamanujan, 2018-10-09:
@paulvha Thanks for your reply
what platform (arduino/ESP8266/ Raspberry / ? ) do you use ?
 - I used Arduino UNO 
which software ?
- Arduino IDE with the below github code
    https://github.com/Seeed-Studio/Mutichannel_Gas_Sensor
what is the VCC (5V or 3.3V)
- I used 5V supply point(on female headers) from Arduino UNO

paulvha, 2018-10-09:
I am currently occupied on another project. I will look later in the week
Paul

sairamanujan, 2018-10-09:
Thank you @paulvha

sairamanujan, 2018-10-09:
@suvodip1212  Thanks for your valuable suggestion, I will try this out and post the results

sairamanujan, 2018-10-10:
@suvodip1212 No improvement in the vaues

Hello community, please help me in resolving the issue

paulvha, 2018-10-14:
took me some time to find the sensor again. What output do you get with the example : RawData.
should be something like
a_ch3 = 860
a_co = 950
a_no2 = 155
5
R0:
295.46	728.77	10.00
Rs:
64.10	642.63	3.43
ratio:
0.22	0.88	6.41

sairamanujan, 2018-10-17:
> took me some time to find the sensor again. What output do you get with the example : RawData.
> should be something like
> a_ch3 = 860
> a_co = 950
> a_no2 = 155
> 5
> R0:
> 295.46	728.77	10.00
> Rs:
> 64.10	642.63	3.43
> ratio:
> 0.22	0.88	6.41

Could you please tell me what are the units of the concentrations. Also please lighten me on the process you followed for getting accurate values.

paulvha, 2018-10-17:
HI
let me try to explain

The sensor has 2 x 3 storage places for R0 CH3/ CO/ NO2 values: Those RO values stored from the factory, those values stored by local/user calibration. Either the factory or user RO values  are then used to compare against the RS values measured in the current environment. RS / RO (either factory or user) gives the ratio. All Rawdata does is read the factory RO values and the current RS values to display and calculate the ratio. By looking at the values, we can see whether your issue is because of the RO value or RS value and help to diagnose /solve the issue.

The 860 / 950 / 155 seems to be factory values in most devices. The RO ADC values ( 295.46, 728.77 and 10) are derived from that. The RS value in the example was the information that was measured at the time I took the sample.

regards, Paul

sairamanujan, 2018-10-27:
@paulvha Thanks for your reply.

I would like to know, how did you know that the values from the sensor are accurate ?

paulvha, 2018-10-27:
I think they are not. In my opinion the board has design issues and I have posted 2 questions/issues: one with respect to the heating power and one with respect to the factory values (890 / 950 and 155). Both questions are unanswered, which is OK. If you are interested, I had posted my project on [(https://github.com/paulvha/multichannel-gas). ]
That said these issues look to be different to the issue you are having as the concentration you get is way out of bounds. Either the factory values are off or the values you currently read or the formula. 

regards,
Paul