Seeed-Studio/Mutichannel_Gas_Sensor #11: Can't get the sensor to work with Adafruit's Feather M0

weirdchaos, 2017-10-03:
I'm trying to use the Multichannel Gas Sensor with an Adafuit Feather M0 board but the sketch get stuck at gas.begin().
I tried to put prints in the library code and it seems that nothing gets returned after the Wire.requestFrom in the get_addr_dta function resulting in an infinite loop to START:

I can get it to work with an ESP8266 board though so I know that the sensor is functioning properly.

I don"t know what to do to make it work?

paulvha, 2017-10-09:
looks that Gas Sensor is not responding to the request to provide information from the register: it does a write request first and then waits for 2 bytes from the sensor with the Wire.requestFrom(). Try to debug with an i2c scanner/tracer for arduino. There are a number around.

lorauser, 2019-02-05:
Did you solve the issue? because i got the same with a different board... :/

naamah75, 2019-04-06:
Same issue with MKR1000

ZwilleSmutje, 2020-02-01:
Same Issue for Seeeduino Cortex-M0+

ZwilleSmutje, 2020-02-01:
in the MultichannelGasSensor.cpp (line 397) it says:

Function name:           changeI2cAddr
 Descriptions:            change I2C address of the slave MCU, and this address **will be stored in EEPROM** of slave MCU

And sadly  "Not all SAMD devices use external eeprom #1322 "....
https://github.com/mysensors/MySensors/issues/1322

ZwilleSmutje, 2020-03-24:
I am sorry to have confused everyone, but I withdraw my post from 01 Feb.

The trick to get the Multichannel-Gas-Sensor to work was ... horrifyingly easy.
You can check the Code where I put a note at the Side.

Here is how it goes:
1) download the Library
2) look into the MultichannelGasSensor.h
you will find this:
`
#ifndef __MUTICHANNELGASSENSOR_H__
#define __MUTICHANNELGASSENSOR_H__

#if defined (ARDUINO_SAMD_ZERO)
	#define _SERIAL SerialUSB
#elif defined(ARDUINO_ARCH_SEEED_STM32F4)
	#define _SERIAL SerialUSB
#elif defined (ARDUINO_ARCH_SAMD)
	#define _SERIAL Serial
#elif defined (ARDUINO_ARCH_AVR)
	#define _SERIAL Serial
#else
	#error "Architecture not matched"
#endif`

3) adapt to your board
In my case I am using a Seeeduino Cortex-M0+ which uses an SAMD processor. So I just needed to 
change line 43 and add a USB. That is because the board has different USB output options so you have to specify which one you plugged your Computer to. Most of the times its the USB connection.

#elif defined (ARDUINO_ARCH_SAMD)
	#define _SERIAL Serial**USB**

brychanthomas, 2021-04-06:
I was having the same problem with an infinite loop to START in get_addr_dta when trying to use it on an STM32F103C "blue pill" board. Rewriting the get_addr_dta functions with while loops and delays like in [this fork](https://github.com/brychanthomas/Multichannel_Gas_Sensor/tree/get_addr_dta_fix) worked for me, I can now call begin and read data from the sensor without getting stuck.

closed 2024-10-15 by baorepo