Seeed-Studio/Mutichannel_Gas_Sensor #13: ADC Factory Values

thecockroach, 2018-01-03:
Hi, anybody can explain to me, where did get this values? The adc analog reference of the uc it's 3v3  or 5v? 

  NH3 = 860
  CO  = 950
  NO2 = 155

Thanks

paulvha, 2018-01-08:
see my question from 27 September 2017 around the same issue. The problem is that the ADC reference is taken directly from the VCC instead of after the 3.3V regulator.I have got the same values as reference. In my earlier post I explain why this can not have been calibrated with 5V.  No answer so far. Also see my other post around the (lack of) heater power. I believe this Multichannel sensor board has 2 serious hardware design errors which makes me doubt about the measured results.
Paulvha

thecockroach, 2018-01-15:
Hello, I bought the Generic Sensor on Ebay, but the values of the heater are calculated with 5V as its datasheet shows.
I want to understand what have been done looking at their design, they have pull ups of 56k also I cannot understand that configuration, I'm in touch with an institute that calibrates instruments in my country (INTI), and I want to go to the laboratory with my circuit and verify with another laboratory parameter.
So far my circuit is going to be powered to 5v, with the resistors suggested by the datasheet, and my arduino UNO will also be 5v.  the firmware I'm using is the one here as "firmware_new", I need to know what the 56k resistors do , what happens if I discard them and go directly to the ADC....
I'm able to provide the data thrown with that configuration by comparing the values with another calibrated probe.
Would you suggest some other changes?
Best Regards, thecockroach

paulvha, 2018-01-16:
Good to see you are going to use 5V also for the heating. The 3.3V regulator that the Seed Studio board is using could be by passed of course. Another important point is the reference voltage to use : AREF (pin 20) has a capacitor, so either the ATMEGA168 is programmed to use AVCC or an internal 1.1V. It is logical, and determined, that AVCC (pin 18) is used which is sourced through a low pass filter (R11/C3) from VCC. Make sure this AVCC is tied to a stable 5V. The more stable the better. 

About the 56K: For the ADC to measure the resistor change of a sensor (CO, N02, NH3), you need to be a voltage bridge between VCC (5V) and ground.  If you do NOT do that, the other side of the sensor will show 5V as there is NO voltage drop. The ADC is measuring  the sensor resistor change as a change in voltage drop.  V= I * R. V= voltage, I = current, R = resistor). With a stable voltage, input VCC, you can only measure the impact of a change in resistance if there is current flowing through the resistors. Hence the voltage bridge. so why 56K?  well because the datasheet on page 3 is mentioning that a single resistor of 56K could do the job. However if you want a different sensitivity that resistor could be as low as 820 ohm. In the datasheet they show a schema of a voltage bridge where one side of the sensor is tight tied to ground and the other side connected to a number of serial resistors. If the IO connections are open, the total resistor value  is : 1.2M + 100K + 6.8K = 1.3068M. However it the top IO is set to 5V, now the resistor value change to 100K + 6.8K = 106.8K. If the bottom IO is set to 5V, the resistor value is 6.8K. There is a good document about this specific topic for sensitivity: https://www.cdiweb.com/datasheets/e2v/Load%20resistance%20for%20MiCS%20sensors%20measurements.pdf. Seems that they advice around 400K.

Keep me posted on your results. looks like a fun project..

regards
Paul

closed 2024-10-15 by baorepo