Seeed-Studio/Mutichannel_Gas_Sensor #14: Working principle of sensor's firmware

Pilius, 2018-01-16:
Hello.

It looks like it is good discussion about working principle of sensor, schematic design and ppm calculation algorithm. 

I would like to get more information about concentration calculation, especially about Ethanol detection with "RED" sensor.

I have found most important parts about calculation in "MutichannelGasSensor.cpp" file.

Let us start from the end, from the result:
1. c = pow(ratio1, -1.552)*1.622; 
This power function makes ppm value from ratio, like in MiCS-6814 [DATASHEET.PDF](https://raw.githubusercontent.com/SeeedDocument/Grove-Multichannel_Gas_Sensor/master/res/MiCS-6814_Datasheet.pdf) , first table. 
2. ratio1 = (float)An_1/(float)A0_1*(1023.0-A0_1)/(1023.0-An_1);
How does this proportion relates to ratio of two resistances? 
3. int A0_1 = get_addr_dta(6, ADDR_USER_ADC_CO);
This looks like it is R0 value stored as unsigned integer. 
4. int An_1 = get_addr_dta(CH_VALUE_CO);
This - Measured resistance, or unsigned integer value of analog reading.

How does firmware of this sensor calculates ratio of resistances Rs/R0 ? With Rs - it clear that voltage drop is read with ADC.

paulvha, 2018-01-16:
If you have read the other posts, you will see that I highly question (with reason) the results of this board. That is not because of the MiCS-6814 (which I believe is very good), but the hardware design on this board around it. The reference voltage and power for the heater are a major concern for the correct readings.  Back to your question...
The resistance is in indication of the pollution. By providing a voltage divider, the ADC reading could be a correct indicator of the resistance change of the sensor. The ratio is based on comparing the resistance R0 (ADC reading) : "under controlled conditions" (for this see page 2 of the datasheet)  to the resistance  (ADC reading) in current air (R1).  R0 is claimed to be detected under controlled conditions and stored . I have serious question around the applied VCC under which this R0 is measured. (next to the power/temperature of the sensor)   R1 is expected to be measured in the same way as R0, but now in current environment. The firmware does NOT calculate the ratio. the sketch does.  All the firmware does is read and communicate the current ADC level back to the sketch. For R0 that was in controlled environment, for R1 that is in current environment.  The assumption is that R0 and R1 are measured by the firmware in the same way, but only in different air.  However, the calibration (R0), can not have been done with 5V... one could only assume 3.3V.... but that is an assumption. The board specification states that the VCC can be between 3.3V and 5V, which has a MASSIVE  impact on ADC reading results.(either R0 or R1) I have explained that in detail in the ODT/ Word file my github documentation :https://github.com/paulvha/multichannel-gas. The MiCS-6814 is a good chip... but in my mind the board has 2 seriously problems in the hardware design (as documented and unanswered questions on this forum) that makes me deeply wonder about the correct results...

Pilius, 2018-01-17:
Hello, paulvha, thank you for your reply, I  really appreciate that.

>If you have read the other posts, you will see that I highly question (with documented reasons) the >results of this board.

Yes, I have read in other issues that it was hard to match  concentration readings from sensors and actual concentration of technical gasses. We have performed ethanol test and we got quet good linearity. Latter I could show our test results.

So, I understand that R0 was measured once in controlled environment, and stored in ATmega chip. R0 value is somewhere between 100 and 1500 kOhm like described in datasheet. 
>Sensing resistance in air R0 is measured under controlled ambient conditions, i.e. synthetic air at 23 >±5°C and 50 ± 10% RH for RED sensor and synthetic air at 23 ±5°C and ≤ 5% RH for OX sensor
What was gasses concentration in the air? Was it 60ppm of CO like described in note 2?

At the moment ADC measurements I am putting aside

paulvha, 2018-01-18:
I have no idea what the gas concentration was in the air at the time of calibrating. One would think it is 60PPM, as what is now called and stored as R0, was RS at the time of doing the calibration. However looking at the CO diagram, the 60PPM concentration is when Rs/R0 = 0.1. So it looks it was about 6PPM. Depending on where you are in the world, there is a different definitions of clean air. Living in the Netherlands, the EU has defined this as 10mg/m3 ( http://ec.europa.eu/environment/air/quality/standards.htm). Translating that to PPM (molecular weight of CO = 28.01 g/mol) using the website http://www.aresok.org/npg/nioshdbs/calc.htm, gives 8.73 ppm. Wikipedia page https://en.wikipedia.org/wiki/National_Ambient_Air_Quality_Standards, comes to the same conclusion. So I expect the calibration was done in even cleaner air :-)

For the ADC reading :
let's take the red sensor as example. Assume it has the lowest 100K. It is connected to the serial resistor 56k. Total resistance 156K. The current flowing:  3.3V/ 156K = 21.15mA. The voltage over the sensor is  21.15mA * 100k = 2.115V. That voltage is read by the ADC. Without the 56K resistor there would be an EXTREMELY low current flowing, as the ADC has a very high impedance value,  and the ADC would read ~3.3V. (ALWAYS). Hence you MUST have a voltage divider 
The ADC is 10 bits. if the reference voltage (VCC in this case) is 3.3V, each step reading is 3.3/ 1024 =0.0032v. The ADC will return a value of 2.115v / 0.0032 = 656. If the reference voltage (VCC) is 5V, each step is  5/1024V =0.00488, the ADC will return in that can 2.115V / 0.00488 = 433.

MarcFinns, 2018-01-27:
anyone interested in collaborating for a better library for this board?

paulvha, 2018-01-27:
The driver could be better, but given the hardware design errors, I don't think it will help. Too bad as the MiCS6814 is a great chip...

MarcFinns, 2018-01-27:
I was puzzle by the strange numbers coming out of this board so i am very interested in your considerations. I found the board schematics. Let me understand your point. 
The default ADC Reference is VCC itself so i see the first issue. This could be addressed by connecting Aref to 3.3V or using the internal 1.1V (too low for the range).
However I see the fact that they probably did not understand the second measurement schematic in the cryptic MiCS datasheet. Probably with some nifty soldering we could add a simple voltage divider but given the range, 10bits would hardly be sufficient. Some of the I/O pins however are still available to fully implement the schematic, but a totally new reading logic must be omplemented in the firmware... worth it? (not many boards with this sensor are available)

paulvha, 2018-01-27:
HI Marc, currently working on another project and i have one waiting. On this moment  I give those 2 priority. Maybe later.. 
regards,
Paul

MarcFinns, 2018-01-27:
i understand. good luck with your projects!

suvodip1212, 2018-01-27:
I would be happy to help. I have ordered the hardware. Soon start the
calibration and then update the code. Keep you guys updated on this thread.

MarcFinns, 2018-01-27:
Please do. I have the impression that Paul might be right, hence some soldering under the microscope might be needed. That I can do, but little time to write firmware and debug it these days!

suvodip1212, 2018-01-27:
A small change that I did in mq sensors for better accuracy was to check
the vcc. It is supposed to be 5 v but rarely is. Here is the small snippet
I used:

void setup()
{
  Serial.begin(9600);
  Serial.println(F("Internal Voltage Sensor"));
}

void loop()
{
  Serial.println(readVcc());
  delay(1000);
}

long readVcc()
{
  long voltage=0;
  uint16_t wADC;

  // Read 1.1V reference against AVcc
  // set the reference to Vcc and the measurement to the internal 1.1V reference
  ADMUX = _BV(REFS0) | _BV(MUX3) | _BV(MUX2) | _BV(MUX1);
  ADCSRA |= _BV(ADEN);  // enable the ADC

  delay(20); // Wait for Vref to settle

  ADCSRA |= _BV(ADSC); // Start conversion
  while (bit_is_set(ADCSRA,ADSC)); // measuring

  // Reading register "ADCW" takes care of how to read ADCL first and ADCH then.
  wADC = ADCW;

  voltage = 1125300L / wADC; // Calculate Vcc (in mV); 1125300 = 1.1*1023*1000
  return voltage; // Vcc in millivolts
}

skanpt, 2019-07-19:
Hi guys, I'm doing a project with MiCS-6814 and I would like to get your help please. As @Pilius asked, what are those values and how do I get them?

//-1.552, 1.622, the expression.. Where does this expression come? I know there's a graph on datasheet but how do you relate it?
c = pow(ratio1, -1.552)*1.622;

//The ratio is Rs/R0? Where does this expression come from?
ratio1 = (float)An_1/(float)A0_1*(1023.0-A0_1)/(1023.0-An_1);

//A0_1 is equal to what? I know where it comes from but what does it represent?
int A0_1 = get_addr_dta(6, ADDR_USER_ADC_CO);

//An_1 is equal to what? I know where it comes from but what does it represent?
int An_1 = get_addr_dta(CH_VALUE_CO);

Sorry for so many repetitive questions but I really need to know that in order to present my project...
Thank you :)

paulvha, 2019-07-22:
Hi

I have not seen an answer , but let me try to help

The graph is the only real reference. What I have done is read as careful as possible the graph and plot the data in an excel graph. Then used the ‘include trend line’ option which gave me nearly the same results. Please see https://github.com/paulvha/multichannel-gas

The multi channel gas meter is comparing the reference values to the current  values. Both are measured with an  ATMega processor which has an ADC which has an accuracy in 1024 steps ( 0 – 1023). Hence you see the 1023 in the expression.  The reference resistor value is mentioned to  be measured under controlled conditions and (also) stored in ADDR_USER_ADC_CO location in the MICS 6814. The current measured resistor value for the sensor  is obtained by reading the CH_VALUE_CO location from the MICS6815. Same is true for the other 2 sensors. Please read the .odt on the earlier mentioned location about serious doubts and potential issues with this sensor.

Regards,
Paul

Brunozwierewicz, 2020-07-07:
Have you used the CJMCU-6814? Can I assume that the libraries for the Grove multi-channel are the same for the CJMCU-6814?

closed 2024-10-15 by baorepo