Seeed-Studio/Mutichannel_Gas_Sensor #22: Does not work with arduino mkr wan 1300 and mkr wifi 1010 probably the mkr lineup...

lorauser, 2019-02-05:
Hey, the problem is that it gets stuck on the gas.begin()...and does not go further... I can see the right address with a i2c scanner, but when I try to read the sensor data nothing happens, The sensor works just fine with arduino uno and nodemcu, and a different sensor(grove eCO2) with the mkr wan 1300 on the same pins and wires works so i know the board is ok... any help?

TwinedGemini, 2019-02-16:
I'm experiencing the same sort of.... it is constantly saying invalid on my mkr nb 1500 (SAMD21 Arcitercture) but is giving values on my UNO... I need it to work on the mkr board though

TwinedGemini, 2019-02-16:
Getting this message:

In file included from C:\Users\Bruger\AppData\Local\Temp\arduino_modified_sketch_284239\ReadSensorValue_Grove.ino:15:0:

D:\Dokumenter\Arduino\libraries\Grove_-_Multichannel_Gas_Sensor/MutichannelGasSensor.h:93:17: warning: inline function 'void MutichannelGasSensor::write_i2c(unsigned char, unsigned char*, unsigned char)' used but never defined [enabled by default]

     inline void write_i2c(unsigned char addr, unsigned char *dta, unsigned char dta_len);

                 ^

naamah75, 2019-04-06:
Same issue as @lorauser ... it hangs on MKR1000 just after `gas.begin()`.
The same module works with Seeeduino Lotus and Wio Node.

pallox, 2019-04-09:
> Getting this message:
>
> In file included from C:\Users\Bruger\AppData\Local\Temp\arduino_modified_sketch_284239\ReadSensorValue_Grove.ino:15:0: [...]

The inline function is supposed to be put in the header file. Otherwise the compiler doesn't know what to insert into the call-sites when it's compiling other source files. It's a problem of the library...

closed 2024-10-15 by baorepo